zafersn/WiFi-RC-Controller-With-Camera #1: I get the error No serial port found

travis4117, 2016-12-20:
I followed the instructions for wifi Pi3 and after uploading the arduino file and running the pi3 image file after boot of the Pi3 I get the error No serial port found.  Maybe it something simple I am a newbie please help.

zafersn, 2016-12-20:
@travis4117  hi. Firstly. Which arduino model do you use?

zafersn, 2016-12-20:
@travis4117   

Now, when we plug OSEPP Nano into raspberry pi, we need to know what USB interface looks like. For example, After connecting the arduino to raspberry pi via usb, the time when we make ls / dev / tty * should appear as "/ dev / ttyUSB0" or "/ dev / ttyACM0". When you make "ls / dev / tty *" on raspberry pi, what is the arduino usb interface? 

Arduino, via usb , while connected to raspberry pi and not connected to raspberry pi . it should look like the image below when we run the command ls / dev / tty * . But You can not do This command ls /dev/tty*, on the .img file you downloaded. İf you want to run this command, you should try it on a normal raspberry pi OS.

https://github.com/zafersn/WiFi-RC-Controller-With-Camera/blob/master/questionImage/noserialportfound1.jpg

zafersn, 2016-12-21:
problem solved. 
http://www.instructables.com/id/RC-CONTROLLER-WITH-CAMERA-V2-PiCAR-V2/

closed 2016-12-21 by zafersn