zafersn/WiFi-RC-Controller-With-Camera #2: Video Stream

DemonXD, 2017-03-14:
Hi, i'm interested in your project, and here is a difficult question for me to show the video stream in a desktop app created by kivy(base to python),just show the image(i guess)?
if you can't show me the specific code, can you tell me the method?

TKS A LOT

zafersn, 2017-03-14:
Hi @DemonXD . To put it plainly, I did not understand anything about the question. Could you write with a little clearer language?

DemonXD, 2017-03-15:
Sorry for that.
i'm writing an app use kivy, control my toy  car(base to raspberry pi 3B), and i also have a Pi camera module on it,so i want to take video stream in my app,just like you did.
but you put it in your android app, i want it in my desktop app.
i search for it a long time, and i can't find some useful method. am i clearly this time? :-)

zafersn, 2017-03-15:
yes This time I get it.)  The only thing I can suggest for you is to investigate gstreamer. Because I am using gstreamer to transfer the image.

For more information about the project or if you have any problems please feel free to contact me!

DemonXD, 2017-03-15:
TKS a lot, i will try it later.
Have a nice day!

sylvie94, 2017-03-22:
Hello,

I really like your project and would like to do it but I was wondering how you connected the boards to the computer to upload your code?
Am I connecting the raspberry Pi or Arduino to the computer and how do I do it?
And do you think it would work with the Arduino Uno instead of the Nano?

Thank you

closed 2017-03-22 by zafersn

karthidigi, 2017-05-08:
Hi zafersn,   I downloaded raspian os for pi 3 and installed it on a 32gb sd card,   
Pi started to boot , after a while  boot screen shows your dhcp is configured and your IP address is 192......,   

pi shows this screen afterwards its not start to show pi desktop.    

I have no idea whether this will show desktop or just stop by here..

zafersn, 2017-05-09:
Hi @karthidigi Yeah. Black screen remains true.There seems to be no error.If you see the name of the network with the name pi car in the wifi list, it means that everything is correct.

karthidigi, 2017-05-09:
I have 16GB and 8gb Sd cards, I Want to write the pi car image to 8 or 16 GB,    how to convert the image file...