zafersn/WiFi-RC-Controller-With-Camera #3: How to upload the code

sylvie94, 2017-03-22:
Hello,
I really like your project and would like to do it but I was wondering how you connected the boards to the computer to upload your code?
Am I connecting the raspberry Pi or Arduino to the computer and how do I do it?
And do you think it would work with the Arduino Uno instead of the Nano?
Thank you

zafersn, 2017-03-22:
Hi. @sylvie94 , thankyou for your question.  Now,
 What version of pi_car 'ın do you want to use? Because version 1 and 2 were removed from the broadcast. You can use the new version (Version 3). All you need to do is download the .img file from the V3 partition and burn your sd card. Version 3 is both faster and safer and also has many features.

closed 2017-03-30 by zafersn

sylvie94, 2017-04-01:
Hello @zafersn 
So I was trying to upload V2 but you say I should upload V3 instead? To do that, I just have to burn the img to the SD card and it should work?
Thank you

zafersn, 2017-04-01:
Hi. @sylvie94 
 Yes, I suggest you install V3. As you say,You just need to write the img file to your sd card.

sylvie94, 2017-04-01:
Ok so if I already have the app installed on my phone will this change anything?

zafersn, 2017-04-01:
@sylvie94  If you have installed the current application on your phone, you do not need to change anything

sylvie94, 2017-04-01:
ok thank you!

sylvie94, 2017-04-01:
I uploaded the img to the SD card and then put it on my raspberry pi but when I turn on my raspberry pi, it just stays stuck at the black screen with [ OK ] started Cleanup of Temporary Directories.

zafersn, 2017-04-01:
@sylvie94 There is no malfunction. It has to look like it. Now, you need to send data from android or windows application for testing.Windows application is free. https://drive.google.com/file/d/0B6yjwSAqPTgfWGlRX3F0T08tYW8/view

sylvie94, 2017-04-01:
oh I cant control it using the mobile app? I need to have the joystick?

zafersn, 2017-04-01:
@sylvie94  No need for joystick for android mobile application. I need a joystick for windows application. Do you see the wifi network called Pi_CAR in the wifi section on your phone. Can you give me a screenshot so I can understand what the error is.

sylvie94, 2017-04-01:
Yes I see the wifi called Pi_CAR on my phone and I was able to connect to it but when I open the application on my phone, I just see a black screen.

sylvie94, 2017-04-01:
[image]

zafersn, 2017-04-01:
beautiful. It looks like everything is normal. Could you connect your raspberry to the screen?

sylvie94, 2017-04-01:
Yes I did and I see a lot of writing hahaha

sylvie94, 2017-04-01:
Now it says 1111111111111

zafersn, 2017-04-01:
:) good job. But  What kind of writing do you see ?

zafersn, 2017-04-01:
please  give me a screenshot

sylvie94, 2017-04-01:
[image]

sylvie94, 2017-04-01:
Why can't I see the streaming on my phone?..

zafersn, 2017-04-01:
As far as I can see, data is coming. Now, enter the settings menu in the android app and enter the speed and pwm values for the vehicle. Example: max speed = 50 pwm ratio = 2

sylvie94, 2017-04-01:
[image]

sylvie94, 2017-04-02:
I changed the speed to 35 and ratio to 1

zafersn, 2017-04-02:
Have you tried button control on the android application?

sylvie94, 2017-04-02:
ohh wow it works thanks but moving doesnt work but I think cuz I changed some stuff in the arduino code.. I will change it back!
Thank you for your help :)

zafersn, 2017-04-02:
:)) YESS. GOOD JOB. But There is a big mistake. It is not used  Arduino in V3. Only raspberry pi

sylvie94, 2017-04-02:
Ohhhh I see so if I change the code in arduino it wont do nothing! Oups ahaha that makes sense now. So do you have the raspberry pi code that I can modify? because I have a different motor for the wheels..

sylvie94, 2017-04-02:
Here is the code I wanted to put in Arduino: 
#include <Servo.h>
Servo left;//Define left servo
Servo right;//Define right servo
byte val;

void setup()
{
  Serial.begin(115200);//Change the baud rate value depending on the default baud rate of your bluetooth module, for Bluesmirf-115200 and for JY-MCU-9600
  left.attach(7, 800, 2200); //left servo on digital pin 9 of arduino
  right.attach(5, 800, 2200); //right servo on digital pin 10 of arduino
  
}

void loop()
{
 int a=0;
 if(Serial.available())
  {
    val=Serial.read();
    Serial.println(int(val));

if(int(val)==49)//Move front
{
  right.write(180);//Rotates servo clockwise
  left.write(0);//Rotates servo anticlockwise
}

if(int(val)==50)//Move back
{
  right.write(0);
  left.write(180);
}

if(int(val)==53)//Stop
{
  right.write(90);//Stops the servos
  left.write(90);//Stops the servos
}

if(int(val)==51)//left
{
  right.write(180);
  left.write(90);
}

if(int(val)==52)//right
{
  right.write(90);
  left.write(0);
}

if(int(val)==55)//Extra button
{
  //Enter your code for any extra commands
}

}
}

sylvie94, 2017-04-02:
Is it possible to make it work?

zafersn, 2017-04-02:
Version 2 is older than version 3 and does not have many features. For example, in V2 there is no camera movement, joystick, vr control.

sylvie94, 2017-04-02:
There is no camera movement but the robot moves no? And I don't need the joystick if I'm using the app?

zafersn, 2017-04-02:
In V2, the camera is fixed. So the camera does not move to the left or right. The vehicle can only be controlled with the buttons.

sylvie94, 2017-04-02:
That's fine is it possible to get it please?

sylvie94, 2017-04-04:
Hey @zafersn can you please give me your v2 so I can try it? You are my last hope! Thanks

sylvie94, 2017-04-07:
Hello @zafersn today I was trying to use the camera with the app and it is not working anymore? I don't know why. Can you help me please?

zafersn, 2017-04-07:
hi @sylvie94.  Of course. I am happy to help you. How do you get an error? it's been working before, now is not he working?  I am asking you to take the camera image by connecting raspberry to the screen. If you can not get an image, can you take the error message on the screen or take the picture off the screen and throw it to me?

sylvie94, 2017-04-07:
So I was able to get the camera working last week thanks to your help but this morning I was trying to open the app and see the camera but it didn't work. The camera light turns on but on the screen I get an error message about no data received.. I will add a picture of my screen

sylvie94, 2017-04-07:
[image]

zafersn, 2017-04-09:
hi @sylvie94 . Did you solve the problem?  or What kind of situation is it now?  I checked the applications and software, and I did not see a problem.

The test screens are as follows.

[!screens](https://github.com/zafersn/WiFi-RC-Controller-With-Camera/blob/master/questionImage/issues_3_%202017-04-09%20at%2016.59.15.jpeg)

sylvie94, 2017-04-09:
Hello, no I haven't solved the problem yet. Do you know what it could be? 
I will try to reformat the SD card. 
Thank you

zafersn, 2017-04-09:
@sylvie94 .yes as you say, you can will try to reformat the SD card. Check the connections of the camera and raspberry

sylvie94, 2017-04-09:
I am still having the same error and I reformatted the SD card