zafersn/WiFi-RC-Controller-With-Camera #4: Question about vehicle typ1 wiring

eddyuhl, 2017-04-04:
Hi,
 
i have a question about the wireing of vehicle V3 Typ 1 with one DC motor for forward and backward and one servo for left and right) . I want to use a L298N motor driver for the DC so why you have 2 pwm connections for the dc - motor drive for left pwm and motor drive for right pwm and also enable pin for left and right motor.  I thought i will have one pin for forward, one pin for backward, and a pwm for speed.

I m a little bit confused how to connect in my type oft car. Witch pin do i have to use?

Thank you very much for this great project and your answer!

zafersn, 2017-04-04:
Hi @eddyuhl . Thank you for the your question. Now. The L298 motor drive has 4 directional pins and 2 pwm pins. For you to use an engine in this project (  **vehicle V3 Typ 1** ).  The input pins you need for l298 are: one pwm pin (For speed control) and two direction pins (for Forward and backward). However, you can also do this. You can control both direction and speed by using 2 pwm pins. The connection shape is as shown in the picture below. Enter 5v to enable pins. Import pwm on raspberry to in1 and in2 inputs. even and even ,You can get 4 ampere for output by shorting **in1 and in3**, **in2 and in4**, ENA and ENB.

 **vehicle V3 Typ 1**

Raspberry pi L298 pin connection: 
[image]
Raspberry pi  4 Amper L298 pin connection:
[image]

eddyuhl, 2017-04-05:
Ok thanks for that quick answer! I'll try this!

eddyuhl, 2017-04-07:
Hello,

I need a little bit help again. I have installed the app and the raspberry image, wired the DC-motor for forward and back like you had drawn. Starting the motor works if i press up and down arrows in the app. But it also starts if i press the left and right button.  Up and left is the same turning direction and down and right. Is this the normal function in the software, because it is also used for a tank vehicle and if I there press left it has to turn the right motor forward while the left turn back?
The second question is, how does your tank vehicle drive back, because i measured all wired gpio's. All enable gpio's (23,24,20,21) are switched on all the time. I do not know the bts7960, but i thought these gpio are used to enable the direction forward and back for each motor and the pwm pin defines the speed (like it is normally used on L298). Actually i do not have an idea how to wire this to get it work like:
pressing up - forward 
down  - back 
left - steering servo left
right - steering servo right

zafersn, 2017-04-07:
Answer to first question:  What type of vehicle did you do? For the first vehicle type, if you only press the right or left button, the vehicle will not move but the steering control servo will turn to the right or left. If you press both the forward and the left or right button at the same time, both forward and rightward will move to the left. Please check the connections if the system is working except this. For the second vehicle type, e.g. If you only press the right button, this will cause the vehicle to turn instantly.  In this sudden turn, the right engines run back and the left engines run forward.

The answer to the second question: 

Yes. All enable gpio's (23,24,20,21) are switched on all the time.  If you want to use L298 in the tank system, you do not need to use these pins. You can apply the system I told you here.
You can apply the system I told you [here.]( https://github.com/zafersn/WiFi-RC-Controller-With-Camera/issues/4#issuecomment-291631022 )

eddyuhl, 2017-04-08:
Hi @Zafersn,

I want to do a vehicle with one DC forward and back and a servo for steering left right. I downloaded the raspberry image from here:
 V3"If you want to manage Pi_CAR via Raspberry pi 2 or the built-in Wi-Fi module on the Raspberry Pi" 
(https://drive.google.com/open?id=0B6yjwSAqPTgfcHRRLXYwMHZ6RWs)
The only pins I wired were gpio 13,19 to IN1and IN2 like you had drawn in your picture. And I have to put in the jumper on L298 ENA otherwise it will not work. It was just for a test. Forward and back worked but the motor also turned when I pressed left and right. Maybe I have the wrong software image if there are different types?

zafersn, 2017-04-09:
hi @eddyuhl  I'll get back to you as soon as I can. I'm reviewing it.

zafersn, 2017-04-09:
hi @eddyuhl. Today I got the opportunity to examine the codes and I did examine it.  Yeah . As you say for vehicle V3 Typ 1. Only when the right button is pressed, the servo does not turn to the right. At the same time the car is moving forward. I do not remember why I did this. Is this a problem for you?

eddyuhl, 2017-04-11:
Hi, it is not a big Problem. but maybe if you find time and see the possibility to correct it, it would be very nice if you could solve this. In my case i do not need it within the next days. Maybe later.

eddyuhl, 2017-04-20:
Hi Zafersn,

I got a new servo an wanted to test it. I connected it to 5V USB Power Pack 2A and GPIO 16 like you had drawn. It is this one 
https://www.amazon.de/gp/product/B01KZKOIZC/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
The servo started moving as soon as I connected it to GPIO 16. It is shaking about 10° in each direction. When I touch the left or right button at the smartphone APP it moves to that direction about 45° and stops shaking. 
What can I do to avoid this?
Did you test it? 
What kind of servo did you use?

Best Regards

zafersn, 2017-05-01:
hi @eddyuhl I am so sorry.  I forgot to answer the question. To avoid this problem, you need to feed the servomotors with a better power supply( The current may be insufficient. ). 
Or  You can install 10uF-100uF capacitors on the gnd and 5v pins of servo motors.
Yes. The system was tested.
A good power supply will solve the problem.

closed 2017-05-03 by zafersn