zafersn/WiFi-RC-Controller-With-Camera #6: Using Arduino Uno

suprojitdas, 2017-05-01:
I would be using Arduino Uno for this project. Will it work with the same code and same technique? If you have any changes that you would suggest, please tell me as soon as possible.

Eagerly waiting for your reply.

zafersn, 2017-05-01:
Hi @suprojitdas  

You do not need to use arduino for version 3. Just use raspberry pi. V2 and V1 have been removed from use.

closed 2017-05-03 by zafersn

karthidigi, 2017-05-08:
Hi zafersn, 

I downloaded raspian os for pi 3 and installed it on a 32gb sd card,
I configured my android static ip and everything..

Pi started to boot , after a while boot screen shows your dhcp is configured and your IP address is 192......,

pi shows this screen afterwards its not start to show pi desktop.

I have no idea whether this will show desktop or just stop by here..    

I haven't downloaded your app yet,  I have to pay for it,  before I pay for purchase the app. I want to configure pi.. 

Kindly help me