zafersn/WiFi-RC-Controller-With-Camera #8: Failed to start Avahi Stack and Hostname services

AWKT, 2017-06-19:
I really like you platform, thanks for developing it.  I installed V3 on a RPi 3 with built in WiFi and a second generation RPi camera module.   I can see the network and connect to it, but I cannot enable video.  I plugged the RPi into a monitor and watched the loading screen, the Avahi mDNS/DNS-SD Stack and Hostname Services failed to load.  I had a keyboard plugged in also, but the system seems to be stuck loading something, so I am not able to access the prompt to enter any commands.  Without being able to type commands, I don't know how to do any troubleshooting.  Do you have any suggestions?
[image]

zafersn, 2017-06-19:
hi @AWKT.  I am sorry to answer late to the question. The first action you made is correct. See what's on screen after connecting raspberry to a screen. According to the above picture.   Wait until the time in the last row of 'A start job is running ... / 1min 33 s' is filled in according to the above picture.Once this period is over, the wifi broadcast must start automatically..After this step, send a command to the raspberries over the phone and observe that the commands you send come to the raspberries. If no command is received, configure your phone wifi.

AWKT, 2017-06-20:
Thanks, I had previously used my phone with the RPi as a hotspot and streamed video, so I had assumed I did not need to configure the phone wifi.  I fixed it tonight and it worked great.  Thanks so much.

zafersn, 2017-06-27:
I'm glad. Good luck.

closed 2017-06-27 by zafersn