zafersn/WiFi-RC-Controller-With-Camera #11: Sharing of codes

crisdeodates, 2017-11-02:
Hi zafersn, 
First of all, great job. We see that you are upgrading yourself along as technology is upgrading itself. I would like to make this concept as my project. I have been trying the same concept of android wifi controlling of car with the video stream back to phone, control it over internet and  attach some couple of extra sensors. We all would be so grateful if you will be able to share the codes (raspberry and android parts) so that we can modify it in our requirement by adding more features.
Thanks

zafersn, 2017-11-02:
Hi @crisdeodates  Thank you for your lovely thought.

I really want to share the code and improve the project. But I do not have enough time to share. and there is a monthly rate of 1-2 downloads, I do not know if you believe it, but I even need that money.

crisdeodates, 2017-11-02:
Hi @zafersn ,

If you are comfortable, can we discuss this matter through email?
cris.thomas1@gmail.com

Thanks

closed 2018-01-14 by zafersn

zafersn, 2018-03-25:
Hi. @crisdeodates  .   I shared source code for raspberry pi. if you still need it you can look.