zafersn/WiFi-RC-Controller-With-Camera #12: SSH Login Credentials - V3 Image

GaltresIT, 2018-02-19:
What is the Username and Password for the V3 image in the Readme file? 

I want to use the Pi for more than this project.

Tried the Raspbian default combination but had no luck. (pi/raspberry)

Thanks

zafersn, 2018-02-22:
Hi. @GaltresIT . Have you tried the instructions above?

teias27, 2018-03-11:
Merhaba, bu projede USB webcam kullanılabilir mi.

zafersn, 2018-03-11:
Merhaba @teias27 usb webcam denemedim acıkcası. Ama bu konuyu birlikte inceleyebiliriz. Calıştığım için malesef şuanda uğraşacak vaktim yok ancak sizin vaktiniz varsa kodları açabilirim ve takıldığınız yerde bana soru sorabilirsiniz?. Kabul ederseniz bunun için yeni bir issues açıp soruyu ordan sormanız ve konuyu ordan devam ettirmemiz daha güzel olur.

closed 2018-03-17 by zafersn