zafersn/WiFi-RC-Controller-With-Camera #13: PIN'ler ve breadboard hakkında

ShnAhmet, 2018-03-17:
1 adet L298N motor sürücü, 1 adet LM2596 güç kaynağı 5v, tank şasi (2 adet dc motor üzerinde mevcut), raspberry pi3 (wifi ve bluetooth üzerinde), 1 adet arducam, 12v lipo batarya, jumper kablolar, 2 adet servo motor....yani her şey tamam görünüyor. Sizin Pi_CAR programı da telefonumda yüklü..Tank modeliniz 6 motorlu, oradan bakıp yapmaya çalıştım ama beceremedim. Rasp ile telefonum arasında bağlantı kurulduğunu düşünüyorum ama joystick ekran modelinizin ana görüntüsü dışında başka bir şey yok. Rasp PIN girişleri ve breadboard ayarlarını gösteren bir şemaya ihtiyacım var. Yardımcı olabilir misiniz?

ShnAhmet, 2018-03-17:
Lipodan LM2576 ya baglanti ayni.breadboarddaki servo motor baglantilari ve rasp baglantisi, servosan raspa baglantilar da ayni.raspin guc kaynagi baglantisi da tamam ama sizdeki 4 pil yani 6v, bendeki ise 6 pil yani 9v. Buraya kadar ok. Bundan sonrasinda olayin sikintili kismi bence: L298N motor surucunun de harici guc kaynagina bagladim.dedigim gibi sizin baglanti semaniz anladigim kadariyla 2 adet bts7960 surucu mantigina gore kurulu bana ise tek bir L298N ile 2 dc motor baglantisini dogru yapmaya calismak. bunun rasp pinlerine giris cikis baglanti şeması. PiCar ile Wifi baglanti kontrolunu ise yarin yapacagim su an ekranda bir sikintim var.

Yani kisaca:
1) raspin guc kaynagi illa 4 pilden mi olusmaliydi?
2) L298N motor surucumu bir harici guc kaynagina baglamam yanlis mi? Ve rasp ile baglanti semasi ne olmali?

zafersn, 2018-03-18:
@ShnAhmet  raspberry 'e 9v veremezsin.  :( 5V Vermek zorundasın. Raspberr' in çalışıuyor mu ? Ekrana taktığında görüntü gelip gelmediğini kontrol et.

Şimdi l298 motor sürücü bağlantını da şöyle yapmalısın. Sende 4 adet pwm girisi var l298 'in üzerinde ki EN - Pinlerini jumpjer ile 5v 'a kısa devre yap VE IN1, IN2, IN3 ,IN4 E raspberry 'nin pwm pinlerini tak yani
GPIO12, GPIO18, GPIO13, GPIO19. Yukarıdaki BTS7960 Motor sürcüsüne taktığım pwm 'leri sen l298 'in IN pinlerine takacaksın.

Birde harici güç kaynağı olursa mutlaka Güç kaynaklarının GND pinlerini ortak yapmalısın.

[image]

ShnAhmet, 2018-03-19:
Dediklerinizi uyguladim. ena ve enb leri 5v lari ile jumperladim. IN1,2,3,4 u sirasiyla gpio 12,18,13,19 pinleriyle birlestirdim. Motor surucuye harici guc kaynagi bagladigim icin surucunun gnd sinden raspin gnd sine baglanti da cektim..
Sonuc: kamera ve servolar calisiyor yani telefonumdan kontrol ediliyor ama dc motorlar hareket etmiyor.

zafersn, 2018-03-20:
DC motor surucun sağlam de mi?  
Kaç V ile besliyorsun surucuyu? 
Voltmetre ile sürücünün çıkışlarını ölçer misin ileri geri hareket tuşlarına basiliyken bakalım çıkış verio mu?
Son olarak in1-2-3-4 e taktiğin pinlerin yerini değiştirerek deneyebilirmisin.?

ShnAhmet, 2018-03-22:
Problem motor surucude çıktı. Su an sistem calisiyor. Tesekkur ederim.

zafersn, 2018-03-22:
Ben teşekkür ederim kolay gelsin, başarılar.

closed 2018-03-22 by zafersn

ShnAhmet, 2018-04-10:
Tekrar selam. Deamon tools ile ".img" uzantili dosyanizi acip kodlara gozatmak istedim ama mumkun olmadi. Gercekten cok merak ediyorum. Kodlariniza nasil ulasabilirim? Ne onerirsiniz?

zafersn, 2018-04-10:
selam @ShnAhmet. öyle kodlara ulaşmazsın. kod dosyasına ulaşabilirsin ama ben oraya kodları .pyc  yani derlenmiş python koduna ulaşabilirsin, oda senin işine yaramaz. Yaklaşık 2Hafta önce raspberry pi için kaynak kodları yayınladım. Buradan ulaşabilirsin.

https://github.com/zafersn/WiFi-RC-Controller-With-Camera/tree/master/T%C3%9CRK%C3%87E/V3/Raspberry%20Pi%20Kaynak%20Kodu

ShnAhmet, 2018-04-10:
Hizli reaktiviten icin tesekkurler..