zafersn/WiFi-RC-Controller-With-Camera #15: V1 and V3

anakjid, 2018-04-18:
About V1, using arduino nano to control dc motor dirive. 
Its mean V3, just used rasperry pi only? for wifi, anndroid apps and motor controller?
Please give me details on the circuit of the project if using only Raspberry pi to do all those work

anakjid, 2018-04-18:
thank you for your feedback. MAy I know if I want to use the V3 method but using DC gear Motor and with the function of V1, is it posibble? I just want to design the wifi RC controller with camera using raspberry pi only without arduino nano. is it possible?

zafersn, 2018-04-19:
In V3 version you can only design wifi RC controller with camera using raspberry. Without Arduino nano. The v3 version is more stable and has more features.

closed 2018-05-23 by zafersn