zafersn/WiFi-RC-Controller-With-Camera #16: This link the OS for Pi 2 Model v1.1 doesn't work

jcbnose, 2018-05-27:
I tried to burn this https://drive.google.com/open?id=0B6yjwSAqPTgfcHRRLXYwMHZ6RWs OS onto the Micro SD but Kernel won't start.  It halted and had mounting issue.  Is this OS compatible with my Pi B or something went wrong during SD card burning?  I burned the card twin, used both OSX and Windows.  Thanks.

jcbnose, 2018-05-27:
Sorry, I didn't use Win32DiskImager to burn but used other burn software.  Not sure that would be the problem.  I will try to download Win32DiskImager and try again.

jcbnose, 2018-05-28:
hi @zafersn  , just burned my 32GB SD card with Win32DiskImager, powered up my rpi B, got more daemons started but got different error - failed to start  "avadi mDNS/DNS SD stack" daemon and got stuck at starting LSB DHCP server.  It halted and couldn't get to login.  Any suggestion?  thanks.

By the way, burning the SD card (15GB of OS) took over 7 hours. Is it normal?  I will use checksum and burn the SD card one more time and see.

zafersn, 2018-05-29:
7 hours? Really? This is not normal.  can you share the output about the error with me? The visual annex on the error can be nice.

jcbnose, 2018-05-29:
After re-downloaded the OS, reformatted the SD card and re-burned, it still failed and stopped at starting LSB DHCP server.  My last two resorts before I give up are 1) buy a new SD card and try again, assuming my current card has bad sector, 2) buy RPI 3 and try again (@zafersn, can you confirm if I use RPI 3 I don't need to burn SD card but just clone the python code in the github?)  Thanks.

zafersn, 2018-05-29:
are you using the recommended wifi modules? the wifi devices tested under[ this row](https://github.com/zafersn/WiFi-RC-Controller-With-Camera/tree/master/ENGLISH/V3#1if-you-want-to-manage-pi_car-via-raspberry-pi-2-or--the-built-in-wi-fi-module-on-the-raspberry-pi-3download-the-operating-system-on-this-link--and-print-to-the-sd-card) were written.

jcbnose, 2018-05-29:
ah....I will try another wifi module that is on the list.

Again, from my previous question, would running on RPI 3 no need to burn SD card?  would it work on RPI 3 Model B+?

jcbnose, 2018-05-29:
Hi @zafersn, just tried it on another wifi module (Realtek 8187L), I experienced the exact same problem.  On the wiki page (https://elinux.org/RPi_USB_Wi-Fi_Adapters), not sure how I can install aircrack-ng.  I can't even boot up.  Any suggestion?

jcbnose, 2018-05-29:
also tried on another Linksys/Cisco wifi module but same problem...

zafersn, 2018-05-29:
@jcbnose you have to download the image file and write to the sd card. because the necessary WiFi configurations and various changes are available. After you install the sd card raspberry, it will boot. Then verify that you see a WiFi network with Pi_CAR ssid li on your mobile phone.

closed 2018-07-07 by zafersn