zafersn/WiFi-RC-Controller-With-Camera #17: V3 img file corrupted ?

pointblank123, 2018-06-26:
I'm fairly new to making projects using Raspberry Pi. I would like to ask, why is it that every time I download the V3 img file and extract it and double click on the img file it says that its corrupted. I have tried to download on 5 separate occasions but each time it still shows the same error. Is this normal and if so should I still just continue to add it to my SD card ?

zafersn, 2018-06-26:
hi @pointblank123  The file is not corrupt. After inserting the SD card into your computer, write the Pi_CAR_V3_full.img on the sd card using the Win32DiskImager program.

pointblank123, 2018-06-28:
Alright so the img file was loaded on the SD card without any issues. Thanks so much for the quick response, however now the app that you have linked to says it is incompatible with my device and I cant buy it. I have an Asus Z008D Zenfone 2 running on an Android 5.0. Will this get solved by using the first method of adjusting my phone settings in Android Studio then I would be able to buy it from the PlayStore or do I need to find another workaround for this ?

pointblank123, 2018-06-30:
So I guess there is no way to get around this ?

pointblank123, 2018-07-05:
No worries about the late reply and thanks for the list. However I don't actually own any of the devices listed so I'm wondering if the code would work with another app and also I cant use the current V3 img you uploaded. My raspberry pi just boots up to the rainbow screen with a lightning bolt. Its not a power supply issue because it can boot up just fine with the most recent version of Raspbian. Is the V3 img using an older version of the OS ? 

P.S- Sorry for the many questions but this is a really interesting project that I would love to be able to complete. Thanks again.

zafersn, 2018-07-05:
If you do not have one of the phones in the list, you can not run the android app. There is a windows application for this project. Have you tried Windows application?

You can run this [python code](https://github.com/zafersn/WiFi-RC-Controller-With-Camera/tree/master/ENGLISH/V3/Source%20Code%20For%20Raspberry%20Pi) in your desired raspberry version. but you need to work on gstreamer libraries to use video, which is a very tiring job.

closed 2018-07-15 by zafersn