zafersn/WiFi-RC-Controller-With-Camera #21: How to create my own application.

azimsyafiq, 2018-09-15:
Is it easy to make my own android application? I have a school project and need to make my own application. But I'm new to this. I know there is no way you are going to give the source code for free. So how do I make one

zafersn, 2018-09-15:
hi @azimsyafiq. If you have some experience with android it is very simple to do a similar control application, but the image transfer part can be a little challenging.  I used the gstreamer libraries and guidelines for video streaming.

azimsyafiq, 2018-09-15:
I am a beginner to this. So it might be challenging for me. So I need to learn how to make the GUI, and image transfer right? Is that all or is there other things. I have no experience with android development, sorry.

zafersn, 2018-09-15:
It can be a bit difficult if you do not have experience in the android development process. In this case, let's start step by step. first move the vehicle's motors without image transmission. To do this, you need to search the UDP / IP protocol on the android and communicate with RaspBerry via Android UDP / IP protocol. [Here]( https://github.com/zafersn/WiFi-RC-Controller-With-Camera/tree/master/ENGLISH/V3/Source%20Code%20For%20Raspberry%20Pi ) are the source codes for raspberry pi:

azimsyafiq, 2018-09-15:
How to move it then since no android application? Or do i need to make one from android studio

zafersn, 2018-09-15:
it would be better to start by finding a source code example on the internet and working on it.

azimsyafiq, 2018-09-15:
Yes, I am searching for that. Thanks for responding to my newbie questions.

closed 2018-10-09 by zafersn