zafersn/WiFi-RC-Controller-With-Camera #22: Why Video Cannot Stream

lencaihensem, 2018-11-04:
Hi, im using V1 the app is connected with rasp but the video do not stream. What i must do?

zafersn, 2018-11-04:
Hi. The V1 version of the android control application has now been removed from the Google Play Market. the documents belonging to this version of repository V1 are kept only for information purposes. All necessary documentation for version V3 is available. For circuit connection diagram, see: https://github.com/zafersn/WiFi-RC-Controller-With-Camera/blob/master/ENGLISH/V3/select_Vehicle_en.md

You can perform a successful video stream and in  the V3 version. With raspberry's open source code, you can customize the project as you wish. 

[WiFi- RC-Controller V3-Raspberry Pi Source Code ](https://github.com/zafersn/WiFi-RC-Controller-With-Camera/tree/master/ENGLISH/V3/Source%20Code%20For%20Raspberry%20Pi)

lencaihensem, 2018-11-04:
is it possible if i want the v2 ?

zafersn, 2018-11-04:
No, Only V3 available .

lencaihensem, 2018-11-04:
okay i use v3 right now but why there is no camera stream

lencaihensem, 2018-11-04:
Sorry i misconfigure wifi. Thanks.It work like a charm now!!! :)