zafersn/WiFi-RC-Controller-With-Camera #23: renkli ekran ve yıldırım simgesi

ahmetkorkmaz82, 2019-03-09:
sd kartı takıp raspberry model 3b+ güç verdiğimde 4 renkli bir splash ekran ve sağ üstte sarı yıldırım simgesi görünüyor ve açılmıyor.

zafersn, 2019-03-11:
Merhaba @ahmetkorkmaz82 .  Kullandığın güç kaynağıyla ilgili olabilir. Tavsiye edilen güç kaynağı 5v 2A 'lik bir güç kaynağı. Sizin güç kaynağı değerleriniz nedir?

ahmetkorkmaz82, 2019-03-11:
Hızlı cevabınız için teşekkürler. Yukarıdaki mesajı yazmadan önce internette araştırdığımda güç kaynağı ile ilgili olabileceğini öğrenmiştim ancak değişik değerdeki güç kaynaklarıyla denememe rağmen çalışmadı. Ayrıca da raspberry pi yi daha önce bu denediğim güç kaynaklarıyla çalıştırmıştım. Hatta daha sonra güncel raspbian işletim sistemini kurduğumda yine çalıştı. Ben de güncel raspbian işletim sisteminde sizin paylaştığınız kaynak kodu çalıştırmaya çalıştım o da olmadı sanırım statik ip vermek gerekiyor. Ya da başka bir şeyi atlıyorum.

zafersn, 2019-03-11:
> Hızlı cevabınız için teşekkürler. Yukarıdaki mesajı yazmadan önce internette araştırdığımda güç kaynağı ile ilgili olabileceğini öğrenmiştim ancak değişik değerdeki güç kaynaklarıyla denememe rağmen [...]

WiFi-RC-Controller-With-Camera Version 3 'ü kullanıyorsunuz değil mi?
https://github.com/zafersn/WiFi-RC-Controller-With-Camera/tree/master/T%C3%9CRK%C3%87E/V3

Emin olmamak la beraber bu konuyla ilgili olarak size akşam dönüş yapmaya çalışacağım. Bu proje yayınladıkdan sonra üzerinden çok uzun bir zaman geçti. Proje Rasbian ' ın  Raspbian Stretch Lite  image ' ı üzerine kurdum diye hatırlıyorum. yani desktop ekranının açılmaması gerekir.

closed 2020-08-26 by zafersn