zafersn/WiFi-RC-Controller-With-Camera #24: Android App

YXCXL, 2019-03-27:
Merhaba Zafer abi öncelikle projenizi çok beğendim kendi android raspberry kontrol uygulamamı yapma niyetindeyim fakat tecrübesizim kaynak yahut yardımcı döküman ihtiyacım var yardımcı olurmusunuz.

zafersn, 2019-03-28:
Merhaba @YXCXL  teşekkür ederim. Elimden geldiğince yardım etmeyi isterim. Bu proje için yardıma ihtiyacın olursa hiç çekinmeden sorabilirsin.

YXCXL, 2019-04-19:
Size ulaşabileceğim herhangi bir platform mevcutmu eposta ile ulaşmayı denedim ama.

zafersn, 2019-04-19:
@YXCXL Buradan ulaşabilirsin. Hem seninle aynı problemi yaşayanlar senin sorularından faydalanmış olur.

YXCXL, 2019-04-19:
Öncelikle imajda kullanılan framework, stream vs kullanılan yazılımlar hakkında bilgi alabilirmiyim projeyi kendim yapıp webview olarak kontrol etmek istiyorum v1 ve v2deki gibi arduino kullanmakta istemiyorum birçok projeyi inceledim fakat ayrıntılı bilgi kaynak edinemedim kitap öneriniz vs varsa almak isterim çok teşekkürler

zafersn, 2019-04-20:
Raspbian Stretch Lite image olarak kullandım [Buradan ulaşabilirsin](https://www.raspberrypi.org/downloads/raspbian/)

raspberry pi için kaynak kodlarını paylaştım: [Burada](https://github.com/zafersn/WiFi-RC-Controller-With-Camera/tree/master/T%C3%9CRK%C3%87E/V3/Raspberry%20Pi%20Kaynak%20Kodu)

Gorüntü aktarımı için Gstreamer kütüphanesini kullandım. Android ile Raspberry pi arasında. 
Kontrol komutlarını(sağa sola dön vs...) aktarmak için udp/ip haberleşmesi var. 
Raspberry Pide codlar python ile Androidde ise java+cpp(NDK) ile yazıldı.
Şİmdilik aklıma gelenler bunlar.

YXCXL, 2019-04-20:
Çok teşekkürler projelerinizde başarılar dilerim.

ugurcandede, 2019-07-19:
> Raspbian Stretch Lite image olarak kullandım [Buradan ulaşabilirsin](https://www.raspberrypi.org/downloads/raspbian/)
>
> raspberry pi için kaynak kodlarını paylaştım: [...]

Gstreamer dan başka kütüphane kullandınız mı? 
Sanırım Python sürümü olarak da 2.7 kullanmışsınız doğru mudur?
Hali hazırda çalışan Raspberry Pi3 cihazımda kodları çalıştırmak istiyorum.

zafersn, 2019-07-22:
@ugurcandede  
sadece Gstreamer yeterli 
Bizim uygulamalarımız için 2.7 kullandım diye hatırlıyorum.
Bun uygulamanın hepsi pi3 de test edildi. Pi3 üzerinde çalıştırabilirsiniz

closed 2020-08-26 by zafersn